realsenseai/librealsense #14242: Frame didn't arrive within 100ms

richardrl, 2025-08-25:
```
Error getting frames from camera ego_global: Frame didn't arrive within 100
Device info: <pyrealsense2.device: D435I (S/N: 243522073271  FW: 5.17.0.10  on USB)>
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/richard/linkage_wrist/linkage_wrist/deploy.py", line 165, in main
    img_obs = perception.get_obs()
  File "/home/richard/linkage_wrist/linkage_wrist/interfaces/perception.py", line 82, in get_obs
    frames = pipeline.wait_for_frames(timeout_ms=100)
RuntimeError: Frame didn't arrive within 100
```

Is there any accepted fix for this issue in 2025?

I saw some thread about running a "hardware reset" code, but this didn't work.

I am using AMD Threadripper 7960x CPU. This frame timeout issue happens 100% of the time after running my script, and is immediately fixed by re-plugging the USB, but it is quite a hassle to do so!

I am using Realsense D435.

Also, increasing the ms has no effect - even with 1000ms it doesn't arrive.

MartyG-RealSense, 2025-08-26:
Hi @richardrl  RealSense cameras are validated for use with Intel and ARM processors.  The cameras can work with AMD Threadripper processors but are not validated for use on them. 

Have you previously installed the librealsense SDK on your AMD computer?  Sometimes the **Frame didn't arrive** error can be caused if there is more than one installation of the SDK on the computer and a removal of the old installations resolves the error.

Can you test first please whether adding a **keep()** line to your program immediately after the wait_for_frames() line helps?

```
frames = pipeline.wait_for_frames(timeout_ms=100)
frames.keep()
```

richardrl, 2025-09-01:
Hey @MartyG-RealSense -
Good news! A fix you provided in another thread seems to work:

```
ctx = rs.context()
devices = ctx.query_devices()
for dev in devices:
    dev.hardware_reset()

print("Sleeping for 3 seconds to allow cameras to finish resetting...")
time.sleep(3.0)
```

However the caveat is you must sleep for some time because the hardware reset to asynchronous.

MartyG-RealSense, 2025-09-01:
It's excellent to hear that you found a solution.  Thanks very much for the update!

MartyG-RealSense, 2025-09-25:
Hi @richardrl  Was the advice in the comment above helpful to you, please?  Thanks!