realsenseai/librealsense #14247: Is Jetpack 6.2 available for Jetson AGX?

becketps, 2025-08-26:
I was wondering that there is no docker container or driver for running the Realsense 435i on a jetpack 6.2 system like the Jetson  AGX system. 

Is there any possibility?

MartyG-RealSense, 2025-09-07:
Hi @becketps  Do you require further assistance with this case, please?  Thanks!

becketps, 2025-09-21:
I think so far not. Do you also provide cameras to test it with half transparent horses? A lot of cameras are only able to record it 40cm away and elsewise NaN pixels occure and the depth map is not usable anymore.

MartyG-RealSense, 2025-09-22:
I am certain that RealSense cameras have not been tested with a half transparent horse.  :)

For semi transparent surfaces in general though (such as a plastic drinks bottle) the D405 camera model (which has an ideal depth range of 7 cm to 50 cm) performs very well, as demonstrated at the link below.

D405 is designed specifically to provide high accuracy, high quality images at close range.

If you need both close range and further distance, a D435 would be the next best choice as it has a minimum depth distance of around 15 cm and a maximum of 10 meters.

Other models in the RealSense 400 Series range such as D455 and D415 typically have to be around 50 cm minimum away, otherwise if the camera is placed closer than that then the depth image starts breaking up.

becketps, 2025-10-06:
Yes thanks. But that means that I need other sensoric to see something in the near and see and feel the deformation of hoses . Do you have there ideas what can help for gripping such objects?

MartyG-RealSense, 2025-10-07:
If you need to detect deformations then you could consider taking a 3D scan of the hose and comparing it to a perfect 3D model template of the hose.  In this way, **deviation analysis** can be done to see if there are deformations.

To do this, software called **Artec Studio** can be used with another program called **SOLIDWORKS** via a plugin called 'Geomagic for SOLIDWORKS'.  The Geomagic plugin enables you to take a 3D model and convert it into CAD details and perform operations such as checking for deviation by comparing the scan to a 3D model.

https://www.artec3d.com/3d-software/solidworks-cad 

***********

If you need a gripper-based solution then you can find resources about gripper projects for detecting deformation that make use of RealSense cameras by googling for the search term **gripper deformation realsense**

MartyG-RealSense, 2025-10-24:
You said a couple of weeks ago that you were asked if you had an NDA when you went to file a private ticket on Zendesk.  Was that one of the RealSense team who asked you that on Zendesk, please?  And if they did, what did you reply to them?

becketps, 2025-11-08:
Your request (24846) has been updated. To add additional comments, reply to this email.

That was the only known message I recieved with the answer of the person, who said I need an NDA. On the following message I did not recieve an answer.

MartyG-RealSense, 2025-11-08:
I have checked your ticket and the last message on it was the question from the RealSense team member asking if you had an NDaA  If you wrote a reply, it is not on there.  

On Zendesk, once you finish writing a message you need to go to the **Submit As Pending** menu drop-down in the bottom corner of the page, click on it and select 'Submit as Pending' from the drop-down menu.  The ticket will then be updated with your new message.  

It looks as though the team member has not responded because they are awaiting a reply to the question about whether you have an NDA and it has not been posted on the discussion.

MartyG-RealSense, 2025-12-06:
Hi @becketps  Have you had a response on your Zendesk ticket yet please if you followed the advice in the comment above to use the Submit As Pending button on the ticket?

MartyG-RealSense, 2025-12-06:
The subject of hose deformation sensing is outside of my experience but I will handle your case myself on this GitHub again and take a fresh look at it next week to see if I can find anything new that might be helpful.  Thanks again for your patience.

becketps, 2025-12-09:
you will not find anything because that is research that is true. When the exchange can only be done by NDA without being sure to get there possible hints it will not be the best idea. I only wanted to see if your cameras theoretically can manage this or if they already struggle tue to lighting conditions or depth failures to deformable objects.

becketps, 2025-12-10:
But this NDA I only can sign for myself our institute will not. I mean does it make sense when you say that it is not your main focus that that would be somehow usefull?

MartyG-RealSense, 2025-12-10:
I'm sorry that I cannot do any more. for you without an institute-backed NDA.  I do very much understand your frustration.

closed 2026-10-23 by MartyG-RealSense