realsenseai/librealsense #14253: Large Depth Discrepancy with D435/D405 in Arm Movement – Is this Expected?

qidihan, 2025-08-27:
I am working with the Intel RealSense D435 and D405 cameras and have encountered a large discrepancy in depth measurements. I conducted tests by placing the camera on a robotic arm and moving the arm. The setup involves recording the camera’s position in the base coordinate system (using the robotic arm’s high-precision readings, with an error margin of 0.01mm), along with RGB-D data.

Here’s a summary of the process:
	1.	I performed coordinate transformations to convert depth data from the camera’s coordinate system to the base coordinate system.
	2.	I then plotted the point cloud after the transformation.
	3.	The camera’s intrinsic calibration parameters and offsets are consistent across tests.

However, after plotting the data, I observed a significant displacement in the depth data — the maximum displacement was as large as 5cm. The object itself was stationary, so I expected minimal variation. The parameters I used were 640x480 resolution at 60fps.

Questions:
   •   Is this level of displacement normal when moving the camera (especially on a robotic arm)?
   •   Does this indicate that depth measurements are inherently unreliable during camera movement, especially with respect to object positions?
   •   Could there be any other factors influencing this significant discrepancy?

I am focusing only on the object itself, and the movement of the camera should not have caused such large errors.

Looking forward to your insights.

MartyG-RealSense, 2025-08-27:
Hi @qidihan  D405 and D435 are equipped with fast 'global' shutters on the depth sensors that can handle depth capture when the camera or objects observed by the camera are moving quickly (as fast as a car on a road).

I would recommend trying using 848x480 as your resolution if possible as that is the optimal resolution for depth accuracy on RealSense 400 Series cameras such as D405 and D435.  If a resolution lower than 848x480 is used then depth accuracy can be negatively impacted.

Are the depth values correct when the arm is stationary?  If they are not then it could indicate that the object has a surface that makes it more difficult for the camera to accurately analyze that surface for depth detail.  This could include a plain surface with low texture detail or no texture, or one that is reflective.

The issue with plain surfaces will especially apply to D405, as unlike D435 it is not equipped with an infrared projector component which can cast a semi-random pattern of invisible dots onto surfaces to aid the camera with depth-sensing that surface.

Also, if the objects are certain shapes and sizes (millimeters small or are tall and thin (like a pen or a flower stem) then the camera will have difficulty accurately depth-sensing them.

qidihan, 2025-08-28:
Thank you for your response! I have a few follow-up questions and clarifications:
	1.	When the robotic arm is stationary, the depth values remain almost unchanged and align perfectly in height, so the depth measurement is consistent.
	2.	I have enabled both RGB and depth alignment. Would this have any effect on the depth accuracy?
	3.	Can we directly use the depth intrinsic interface to read the camera’s internal parameters, or do we need to calibrate them ourselves?
	4.	Does the D435 have different versions (old vs. new)? My camera was purchased five years ago.
	5.	The object we are measuring is approximately 11cm x 5cm x 2cm. Could you confirm if the depth measurement will still be accurate for this size?

Lastly, could you provide information on the general depth accuracy we can expect? Specifically, is it possible to achieve a 3-5mm accuracy even when the camera is in motion, and does the speed of the camera movement affect this accuracy?

Thank you for your help!

MartyG-RealSense, 2025-08-28:
D435 has a slower 'rolling' type shutter on its RGB sensor than the fast 'global' type shutter on the depth sensor, so when fast motion occurs the RGB may lag behind the depth image.  A solution for this is to set the depth stream to 30 FPS and the RGB stream to 60 FPS so that their performance becomes equalized and lag / blur during fast motion is much less likely to occur. 

D405 does not experience this RGB lag as all its shutters are the fast global type.

On D405 depth and color both come from the depth sensor as D405 is not equipped with a dedicated RGB sensor.  Some find that they achieve a more accurate RGB image from D405 if instead of using the RGB stream, they use the alternative RGB image provided by the 'RGB from left infrared sensor' mode that D405 supports (D435 does not support this mode).  To use this mode, you use a cfg camera configuration instruction to enable an **infrared** stream instead of the RGB stream and set the infrared stream's format to RGB8 or BGR8 instead of the usual Y8 format that infrared uses.

The D435 model has remained pretty much the same throughout its history since it launched at the end of 2017.  There have been further models based on D435 since then - the D435i (equipped with a motion IMU component), the D435f (a D435 with a light-blocking filter on its lenses that improves depth performance in strong light) and D435if (D435 with both the filter and IMU).

11 x 5 x 2 cm should be accurately detectable, yes.

There is not an official statement for the accuracy of a D435 camera in terms of millimeters.  I once saw advice though that asserted that at 1 meter distance an accuracy drift of 2.5 mm to 5 mm could be expected.  Error should reduce as the distance between the camera and the observed object is reduced.

The chart below also estimates how the depth error on a D435 (the brown line on the chart) will increase as distance from the camera increases, with error being at around zero at the front glass of the camera.

[image]

The D405 camera model is designed specifically for very close range depth sensing and so should perform better than D435 at ranges such as 50 cm / 0.5 m and under.  It is the only model in the 400 Series camera range officially described as having 'sub-millimeter' accuracy.  As mentioned earlier though, depth image quality can reduce compared to D435 when observing plain low-texture surfaces with it.

You can read the intrinsics directly from the camera, yes.

rooftop88, 2025-08-28:
Hi, I tried your suggested setting with **848×400 @ 30 FPS**, and also applied

```python
self.temporal_filter.set_option(rs.option.filter_smooth_alpha, 0.1)
```

I didn’t change any other options. Then I moved the robot arm and collected data to build an ROI. Here are some of the results I observed:

```
Frame 0: ROI mean_base_z=0.0847 m
Frame 0: ROI mean_base_z=0.0722 m
Frame 30: ROI mean_base_z=0.0711 m
Frame 60: ROI mean_base_z=0.0699 m
...
Frame 1110: ROI mean_base_z=0.0820 m
```

Summary:

* Accumulated points: 943,661 (crop 200×180, per-frame colors, no mode filtering)
* Center patch base-frame Z stats (meters):

  * Frames: 38
  * Mean mean: 0.0703, std: 0.0058, range: \[0.0572, 0.0820] @ base\_z
  * Delta to first: mean -0.0019, std 0.0058, range \[-0.0149, 0.0098]

It seems the depth is still fluctuating (around ±1.5 cm). Do you have any suggestions on how we could make the depth more **stable and smoother**? We don’t need absolute accuracy, but we’d like to achieve better **relative stability**.
Below is the result that I plot, it looks very random

[image]

[image]

Thanks!

MartyG-RealSense, 2025-08-28:
If you are using Python code then you could try setting the **High Accuracy** camera configuration preset using code at https://github.com/IntelRealSense/librealsense/issues/2577#issuecomment-432137634

The High Accuracy preset will exclude from the depth image depth values that it does not have confidence in regarding their accuracy.

Whilst setting Filter Smooth Alpha to 0.1 reduces depth fluctuations, it is best suited to a slow moving or stationary camera as reducing this setting slows down the updating of the depth image.  So when moving the camera fast, you may get 'tidal waves' on the image as the image lags behind in updating the depth values to the camera's latest position. 

Another way to control the strictness applied to depth accuracy is to adjust the **Second Peak Threshold** setting (also known as Second Peak Delta), as described at https://github.com/IntelRealSense/librealsense/issues/10224#issuecomment-1060412603

rooftop88, 2025-08-28:
[image]

sunward21, 2025-09-01:
Hello Intel RealSense team,

I am writing to provide some experimental data that corroborates the dynamic depth error issue described in this thread, and to ask about the feasibility of a potential solution.

​​Experimental Data:​​

I conducted a test similar to the OP's setup (camera on a moving robotic arm observing a static scene).   Instead of looking at the full point cloud transformation, I monitored the depth value of a small, stable patch at the center of the image throughout the movement.

I've attached a plot of the results 
[image]

The ​​blue line​​ shows the absolute depth value (in mm) of the patch for each frame.

The ​​orange line​​ shows the delta of the current depth value compared to the first frame.

​​Key Observation:​​

While the depth is generally stable, we observe significant ​​spikes in error (up to 10mm)​​ at specific points during the motion (e.g., near frame 200 and 800).   These spikes are brief and correlate with periods of acceleration or movement change, strongly suggesting they are caused by ​​motion-induced artifacts​​ like motion blur, rather than a slow drift or calibration issue.

​​My Question ：​​

The current DynamicCalibratortool and the rs-depth-qualitytool seem primarily focused on ​​static​​ calibration and quality assessment under ideal conditions.

Given that the D435/D405 cameras have a built-in IMU that can precisely measure acceleration, would it be technically feasible to develop a ​​dynamic calibration or correction mode​​? Or  the current DynamicCalibratortool and the rs-depth-qualitytool can make it
？

This would be incredibly valuable for robotic applications where the camera is in motion.   Could the existing tools be extended to support such a dynamic characterization?   Or is this functionality already available in a less-documented form?

Best regards

sunward21, 2025-09-04:
I DO have an D435i，would it be technically feasible to develop a ​​dynamic calibration or correction mode​​ based on IMU? Or the current DynamicCalibratortool and the rs-depth-qualitytool can make it？This would be incredibly valuable for robotic applications where the camera is in motion. Could the existing tools be extended to support such a dynamic characterization? Or is this functionality already available in a less-documented form?

sunward21, 2025-09-05:
I conducted the same experiment as @rooftop88 ，working with the Intel RealSense D435 and D405 cameras and have encountered a large discrepancy in depth measurements. I conducted tests by placing the camera on a robotic arm and moving the arm. The setup involves recording the camera’s position in the base coordinate system (using the robotic arm’s high-precision readings, with an error margin of 0.01mm), along with RGB-D data.

Here’s a summary of the process:
1. I performed coordinate transformations to convert depth data from the camera’s coordinate system to the base coordinate system.
2. I then plotted the point cloud after the transformation.
3. The camera’s intrinsic calibration parameters and offsets are consistent across tests.

However, after plotting the data, I observed a significant displacement in the depth data — the maximum displacement was as large as 5cm.

Later I found out that we use depth stream to draw point clouds, but the TCP offset (tool coordinate system to camera conversion) used is RGB. Could this be a problem? Should I multiply it by the RGB to depth conversion, using `rs-enumerate-devices -c`? If aligned depth images are used, are both the intrinsic parameters and geometric transformations the same as RGB?

yujunsun110, 2025-09-05:
> D435 has a slower 'rolling' type shutter on its RGB sensor than the fast 'global' type shutter on the depth sensor, so when fast motion occurs the RGB may lag behind the depth image. A solution for this is to set the depth stream to 30 FPS and the RGB stream to 60 FPS so that their performance becomes equalized and lag / blur during fast motion is much less likely to occur.
> 
> D405 does not experience this RGB lag as all its shutters are the fast global type.
> 
> On D405 depth and color both come from the depth sensor as D405 is not equipped with a dedicated RGB sensor. Some find that they achieve a more accurate RGB image from D405 if instead of using the RGB stream, they use the alternative RGB image provided by the 'RGB from left infrared sensor' mode that D405 supports (D435 does not support this mode). To use this mode, you use a cfg camera configuration instruction to enable an **infrared** stream instead of the RGB stream and set the infrared stream's format to RGB8 or BGR8 instead of the usual Y8 format that infrared uses.
> 
> The D435 model has remained pretty much the same throughout its history since it launched at the end of 2017. There have been further models based on D435 since then - the D435i (equipped with a motion IMU component), the D435f (a D435 with a light-blocking filter on its lenses that improves depth performance in strong light) and D435if (D435 with both the filter and IMU).
> 
> 11 x 5 x 2 cm should be accurately detectable, yes.
> 
> There is not an official statement for the accuracy of a D435 camera in terms of millimeters. I once saw advice though that asserted that at 1 meter distance an accuracy drift of 2.5 mm to 5 mm could be expected. Error should reduce as the distance between the camera and the observed object is reduced.
> 
> The chart below also estimates how the depth error on a D435 (the brown line on the chart) will increase as distance from the camera increases, with error being at around zero at the front glass of the camera.
> 
> <img alt="Image" width="964" height="643" src="https://private-user-images.githubusercontent.com/41145062/483055133-c66c9229-3022-41f0-baf9-d3c09a44ba55.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTcwNzYxOTQsIm5iZiI6MTc1NzA3NTg5NCwicGF0aCI6Ii80MTE0NTA2Mi80ODMwNTUxMzMtYzY2YzkyMjktMzAyMi00MWYwLWJhZjktZDNjMDlhNDRiYTU1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTA1VDEyMzgxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM1Zjk5YTUyNmQyOTE2MWE3OTM4ODc4YWY3ZGRiN2FhZGNmMzA1ZWFmNzNlNDEyY2Y0ODM1ZTFmZGExZGNhNDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.1geNjbfmM5ZdOc0ynNVIzxIClkUZTpMOEizG7h9iizs">
> The D405 camera model is designed specifically for very close range depth sensing and so should perform better than D435 at ranges such as 50 cm / 0.5 m and under. It is the only model in the 400 Series camera range officially described as having 'sub-millimeter' accuracy. As mentioned earlier though, depth image quality can reduce compared to D435 when observing plain low-texture surfaces with it. For such surfaces I recommend using a small USB 'visible pattern' projector to add texture to the surface, like the one described at [#13036 (comment)](https://github.com/IntelRealSense/librealsense/issues/13036#issuecomment-2242839115)
> 
> You can read the intrinsics directly from the camera, yes.

I’m curious about how this curve is obtained. I’m currently working with a D435i and trying to create a LUT that maps depth to a parameter k, defined as the ratio between the actual depth and the measured (distorted) depth. My idea is that by multiplying the collected depth values by k, I could correct them to get more accurate depth estimates. However, right now I’m stuck on how to actually build this LUT. Can someone help?

sunward21, 2025-09-08:
> If the **align_to** instruction is used to perform the depth to RGB color alignment then you should not need to program a conversion yourself, as the SDK's **align processing block** software component automatically carries out the calculations necessary to map depth coordinates onto RGB color coordinates and take account of any differences between depth and color (such as D435 having a smaller color field of view size than the depth FOV size).
> 
> The color intrinsic parameters should be used, as after alignment the origin of depth changes from the left-side IR sensor to the RGB sensor. On the D405 model though, which does not have a separate RGB sensor, both the depth and RGB images originate from the left IR sensor by default.

Thank you for the response. What I want to know is if using the align_to instruction, the depth image is used to align the rgb image, or vice versa?
There's also a question about whether using align_to will affect the quality and accuracy of the depth image. If I don't align and instead add the tcp offset from rgb to the rgb->depth conversion, what problems would this cause? Or is this a better choice?

sunward21, 2025-09-09:
Thank you for your reply. I have another question regarding the various coordinate systems of the D435i camera, such as the RGB coordinate system, IR coordinate system, depth coordinate system, etc. Are there any similar diagrams available? I would like to know if there are any visualizations showing the relationships between these various camera coordinate systems. I need to know whether the depth image is based on the left IR camera or another reference method.

closed 2025-09-28 by MartyG-RealSense