realsenseai/librealsense #14260: pyrealsense2: How to find the laser state per frame when emitter_on_off is on?

martinakos, 2025-08-28:
I can set the emitter_on_off in my D435i with this code: 

```
depth_sensor = profile.get_device().first_depth_sensor()
depth_sensor.set_option(rs.option.emitter_enabled, 1)
depth_sensor.set_option(rs.option.emitter_on_off, 1)
```
If I visualise the infra images I get with

`frames = pipeline.wait_for_frames()
`

I can see the laser alternating.  
How can I determine when the laser is on or off with pyrealsense2?

MartyG-RealSense, 2025-08-29:
Here is an example of code for retrieving the laser power metadata in pyrealsense2 instead of ROS:

`emitter = rs_frame.get_frame_metadata(rs.frame_metadata_value.frame_laser_power_mode)`

When the emitter is off, laser power should be 0.

martinakos, 2025-08-29:
That's the C++ syntax. I've been trying the python syntax and I can see the values for laser_power and laser_power_mode alternating 

```
        laser_power = rs.frame.get_frame_metadata(frames, rs.frame_metadata_value.frame_laser_power)
        laser_power_mode = rs.frame.get_frame_metadata(frames, rs.frame_metadata_value.frame_laser_power_mode)
```

There is a weird thing happening though. When I get a laser_power!=0, or laser_power_on==True the corresponding frame infra image should have laser pattern visible but it doesn't. And conversely, when I get a laser_power==0, or laser_power_on==False the corresponding frame infra image shouldn't have the laser pattern visible but it does. 
They are reversed :'(  is this a bug or a feature?

MartyG-RealSense, 2025-08-29:
It is commonly reported by RealSense users that when checking the frame images when the emitter is alternating, the pattern is visible on the image when it is expected to be off, and vice versa.  I believe that in this situation, the metadata value will be more trustable than the image regarding the actual status of the emitter on a specific frame.

MartyG-RealSense, 2025-08-29:
It's great to hear that you achieved a solution - thanks very much for sharing the details!