realsenseai/librealsense #14261: Is there any way to get reduce the depth value fluctuation to 1 mm on D455 Depth Camera (object 1.5 meter distance from camera)

SanuKM, 2025-08-29:
Hi, 
Object(flat surface) is Place at a distance 1.5 meter from the D455 camera. I am trying to reduce the fluctuation in Depth values to ±1 mm.
I want know more on Advance control/mode setting or any other camera parameter setting or any other method which can archive ±1 mm Stability @ 1.5 m

Waiting for the response. 
Thank you

MartyG-RealSense, 2025-08-29:
Hi @SanuKM   If you are using the RealSense Viewer tool then I recommend going to the **Stereo Module > Post-Processing** section of the options side-panel and expanding open the list of post-processing filters.  Then expand open the settings for the **Temporal Filter** and change its 'Filter Smooth Alpha' setting from the default '0.4' to a lower value of '0.1'.  This should reduce depth fluctuations.

SanuKM, 2025-09-01:
Hi @MartyG-RealSense , Thanks for your response. yes i got better results by changing the Filter Smooth Alpha value to 0.1. **But  fluctuation of Depth values is more than ±1 mm at object (flat surface) keep at 1.5 meter way from camera.**  i am using python - pyrealsense2. can you check and advice for me getting this fluctuation of Depth values less than or equal to ±1 mm.

closed 2025-09-14 by MartyG-RealSense