realsenseai/librealsense #14272: RealSense L515 detected by USB but not recognized by SDK or realsense-viewer on Ubuntu 22.04

KimchiPizza, 2025-09-04:
Hello,

I am currently using the Intel RealSense L515 on Ubuntu 22.04. My Linux PC is Jeston Orin NX.
While the camera is detected at the USB port level, it is not recognized by the librealsense SDK.

Both realsense-viewer and rs-enumerate-devices fail to detect the camera.

I have tested with the following SDK versions, but the issue persists across all:

2.54.2

2.50.0

2.44.0

I also updated the camera firmware to the latest version just in case, but the detection problem remains unresolved.

Could anyone kindly advise what might be causing this and how to fix it?

@MartyG-RealSense , your help would be greatly appreciated!

MartyG-RealSense, 2025-09-04:
Hi @KimchiPizza  I recommend using your L515 with the 2.50.0, as that was the final librealsense version that officially supported L515 (it is a retired camera model). 2.44.0 should also be fine though if that is the SDK version that you currently have installed.

Could you first try unplugging the micro-sized end of the USB cable from the base of the L515 camera, reversing the connector's orientation and plugging it back into the camera, please (USB-C cables are two-way insertion at the micro-sized end). Does doing so make any difference?

If reversing the connector does not make a difference, is the camera detected if you launch programs in **sudo** admin permissions mode with commands such as `sudo realsense-viewer` and `sudo rs-enumerate-devices`

If the camera is not detected in sudo mode, may I ask if you are using JetPack 6 on your Jetson.  If you are then this can cause camera detection problems with RealSense camera models that are equipped with an IMU component, like L515 is.  

If you are using JetPack 6 then a solution that works for most cases of camera non-detection because of JP6 is to build the librealsense SDK from source code using the **libuvc backend** installation method, which provides a pre-made build script called **libuvc_installation.sh**

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

KimchiPizza, 2025-09-05:
@MartyG-RealSense , I resolved this issue on another computer(not Jetson) using SDK 2.50.0. Thank you very much for your help.