realsenseai/librealsense #14278: Device or Resource Busy when running D435 and D455 simultaneously on Jetson Orin NX

Frank-mirle, 2025-09-10:
**Hardware**
NVIDIA Jetson Orin NX
RealSense cameras: D435 and D455

**Software**
Ubuntu 22.04
ROS2 Humble
JetPack 36.3.0
RealSense ROS v4.56.4
Built with Librealsense v2.56.4, 
running with Librealsense v2.56.4

**Issue**
I recently switched to a new Jetson Orin NX. When launching both the D435 and D455 simultaneously, I get a “device or resource busy” error.
On this new Jetson, I must specify the camera serial numbers to launch both cameras successfully.
However, on my previous Jetson Orin NX, I could launch both cameras without specifying the serial numbers.

**Launch commands.**
Without serial numbers:
```
ros2 launch realsense2_camera rs_multi_camera_launch.py \
  camera_name1:=D435 camera_name2:=D455 \
  rgb_camera1.color_profile:=1280x720x30 depth_module1.depth_profile:=848x480x30 \
  rgb_camera2.color_profile:=1280x720x30 depth_module2.depth_profile:=848x480x30 \
  initial_reset:=true
```

With serial numbers: 
(To protect privacy, part of the serial number has been masked with x.)
```
ros2 launch realsense2_camera rs_multi_camera_launch.py \
  camera_name1:=D435 serial_no1:="'3325xxxxxxxx'" \
  camera_name2:=D455 serial_no2:="'3365xxxxxxxx'" \
  rgb_camera1.color_profile:=1280x720x30 depth_module1.depth_profile:=848x480x30 \
  rgb_camera2.color_profile:=1280x720x30 depth_module2.depth_profile:=848x480x30 \
  initial_reset:=true
```

**Launch Information**
```
[realsense2_camera_node-2] [INFO] [1757465707.082784961] [camera2.D455]: Device with physical ID /sys/devices/platform/bus@0/3610000.usb/usb1/1-3/1-3.3/1-3.3:1.0/video4linux/video6 was found.
[realsense2_camera_node-2] [INFO] [1757465707.082819522] [camera2.D455]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-2] [INFO] [1757465707.083242514] [camera2.D455]: Device with port number 1-3.3 was found.
[realsense2_camera_node-2] [INFO] [1757465707.083280371] [camera2.D455]: Device USB type: 2.1
[realsense2_camera_node-2] [WARN] [1757465707.083305780] [camera2.D455]: Device 332522073265 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-2] [INFO] [1757465707.083456249] [camera2.D455]: getParameters...
[realsense2_camera_node-1] [INFO] [1757465707.084632164] [camera1.D435]: Device with serial number 3325xxxxxxxx was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1757465707.084846251] [camera1.D435]: Device with physical ID /sys/devices/platform/bus@0/3610000.usb/usb1/1-3/1-3.3/1-3.3:1.0/video4linux/video6 was found.
[realsense2_camera_node-1] [INFO] [1757465707.084988721] [camera1.D435]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1757465707.086318848] [camera1.D435]: Device with port number 1-3.3 was found.
[realsense2_camera_node-1] [INFO] [1757465707.086404356] [camera1.D435]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1757465707.086516552] [camera1.D435]: Device 332522073265 is connected using a 2.1 port. Reduced performance is expected.
```

**Error message**
(Extract a portion of the error message)
```
[camera1.D435]: ./src/rs_node_setup.cpp:511:An exception has been thrown: xioctl(VIDIOC_S_FMT) failed, errno=16 Last Error: Device or resource busy
[camera1.D435]: The device has been disconnected!
[camera1.D435]: Error starting device: null pointer passed for argument "dev"
```

Frank-mirle, 2025-09-16:
Hi @MartyG-RealSense 

This week I will update the new NX to 6.2 and reinstall the SDK along with the related RealSense ROS2 files.

We are using three different brands of NX in total. Some can run two cameras simultaneously, while others require serial numbers to do so. We are not sure whether this is due to a USB hardware issue or an SDK/RealSense ROS2 installation issue.

MartyG-RealSense, 2025-09-16:
Okay, thanks very much for the information.  I look forward to your next report after your NX update.  Good luck!