realsenseai/librealsense #14302: Persistent segmentation faults on Apple Silicon M2 Mac - D455C crashes in HID initialization despite 2.57.3 fixes

jshaw, 2025-09-19:
## RealSense D455C Debugging Report - Apple Silicon M2 MacBook Pro

After extensive debugging across multiple SDK versions (source-built 2.56.5 and Homebrew 2.57.3), the RealSense D455C consistently crashes with segmentation faults on Apple Silicon M2 MacBook Pro running macOS 14.6.1. The camera is properly detected by macOS and works in native apps, but all RealSense tools crash during HID device initialization for the motion sensors. This occurs despite applying recommended Apple Silicon fixes, disabling SIP, and attempting various permission/build configurations.

---

## The big "Question":

Has anyone successfully gotten any version working on  Apple 'M' chips???

---

## Everything I tried is listed below...

## System Information
- **Hardware**: MacBook Pro M2 (2023)
- **OS**: macOS 14.6.1 (23G93) 
- **Camera**: Intel RealSense D455C
- **Goal**: Get depth data working in Processing 4 for people segmentation

## Camera Hardware Detection
✅ **Camera properly detected by macOS:**
```
Intel(R) RealSense(TM) Depth Camera 455:
  Product ID: 0x0b5c
  Vendor ID: 0x8086 (Intel Corporation)
  Version: 50.f1
  Serial Number: 103223060553
  Speed: Up to 5 Gb/s
  Current Available (mA): 900
  Current Required (mA): 720
```

✅ **Camera works in native macOS apps** (PhotoBooth, etc.)

## System Modifications Attempted

### 1. System Integrity Protection (SIP)
- **Disabled SIP** via Recovery Mode using `csrutil disable`
- **Status confirmed**: `System Integrity Protection status: disabled`

### 2. Camera Service Management
Attempted to disable macOS camera services blocking access:
```bash
# Killed camera-related processes
sudo killall -9 UVCAssistant VDCAssistant appleh13camerad cameracaptured
killall -9 ControlCenter

# Attempted to disable launch agents
launchctl bootout "gui/${uid}" com.apple.cmio.ContinuityCaptureAgent
launchctl bootout "gui/${uid}" com.apple.videoconference.camera  
launchctl bootout "gui/${uid}" com.apple.cmio.LaunchCMIOUserExtensionsAgent

# Attempted system-level service disabling
sudo launchctl disable system/com.apple.cmio.uvcassistantextension
```

**Result**: UVCAssistant process immediately restarts after being killed, indicating persistent system-level camera management conflicts.

## Build Attempts

### 1. Source Build - Initial Attempt
```bash
cmake .. -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=false -DBUILD_WITH_OPENGL=false -DFORCE_RSUSB_BACKEND=true -DCMAKE_OSX_ARCHITECTURES=arm64
make -j4
```

**Result**: Consistent segmentation faults in all built tools.

### 2. CMake Configuration Fix for Apple Silicon
**Problem identified**: Incompatible ARM compiler flags in `/CMake/unix_config.cmake`

**Original problematic code**:
```cmake
if(${MACHINE} MATCHES "arm-*")
    set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -mfpu=neon -mfloat-abi=hard -ftree-vectorize -latomic")
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -mfpu=neon -mfloat-abi=hard -ftree-vectorize -latomic")
```

**Fixed configuration**:
```cmake
if(APPLE AND ${MACHINE} MATCHES "arm64-*")
    # Apple Silicon - no special flags needed
    set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -ftree-vectorize")
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -ftree-vectorize")
elseif(${MACHINE} MATCHES "arm-*")
    # Linux ARM 32-bit
    set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -mfpu=neon -mfloat-abi=hard -ftree-vectorize -latomic")
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -mfpu=neon -mfloat-abi=hard -ftree-vectorize -latomic")
```

**Result**: Build completed successfully, but runtime segfaults persisted.

### 3. Library Linking Issues
**Problem**: Runtime library path resolution failures
```
./Release/rs-enumerate-devices:
    @rpath/librealsense2.2.56.dylib (compatibility version 2.56.0, current version 2.56.5)
```

**Fixes applied**:
```bash
# Fixed executable library paths
install_name_tool -change @rpath/librealsense2.2.56.dylib @loader_path/librealsense2.2.56.dylib ./Release/rs-enumerate-devices

# Fixed dylib self-reference
install_name_tool -id @loader_path/librealsense2.2.56.dylib ./Release/librealsense2.2.56.5.dylib
```

**Result**: Library loading succeeded, but segfaults continued.

## Crash Analysis

### Crash Location Identified
**Crash report analysis revealed**:
- **Exception**: `EXC_BAD_ACCESS` at address `0x18` (null pointer dereference)
- **Crash point**: `IOHIDDeviceOpen` in HID initialization
- **Stack trace**: Crash occurs in motion sensor (IMU) initialization:
  ```
  librealsense::d400_motion::create_hid_device
  → librealsense::platform::rs_hid_device::rs_hid_device
  → hid_open → IOHIDDeviceOpen
  ```

**Root cause**: macOS blocks access to HID (Human Interface Device) interface for camera's motion sensors (accelerometer/gyroscope), causing null pointer crash.

### Permission Escalation Attempts
- **Tested with sudo**: Still segfaults
- **Added Input Monitoring permissions**: No improvement
- **Disabled SIP**: No improvement

## Homebrew Installation Testing
**Version tested**: 2.57.3 (latest available)

**Results**:
```bash
# Both regular and elevated permissions fail
/opt/homebrew/Cellar/librealsense/2.57.3/bin/rs-enumerate-devices
# Result: segmentation fault

sudo /opt/homebrew/Cellar/librealsense/2.57.3/bin/realsense-viewer  
# Result: segmentation fault
```

**Identical crash behavior** across both source-built (2.56.5) and Homebrew (2.57.3) versions.

## Processing 4 Integration Attempt
**Error encountered**:
```
UnsatisfiedLinkError: no jnirealsense2 in java.library.path
```

**Analysis**: Processing RealSense library depends on the same native SDK experiencing segfaults, making Java integration impossible.

## Summary of Technical Findings

### Working Components
✅ USB device detection and enumeration  
✅ macOS camera service recognition  
✅ Library compilation and linking  
✅ Basic SDK initialization  

### Failing Components  
❌ HID device access for motion sensors  
❌ Camera power state management  
❌ Device enumeration completion  
❌ All RealSense SDK tools and viewers  
❌ Processing/Java integration  

### Root Cause Analysis
The core issue appears to be **macOS security restrictions** introduced in recent versions that prevent libusb from claiming exclusive access to camera devices, particularly the HID interfaces used for motion sensors. This aligns with:

1. **Apple's USB policy changes** mentioned in Intel engineer comments
2. **UVC driver conflicts** where macOS claims camera as standard webcam
3. **HID security restrictions** treating motion sensors like input devices

### Versions Tested
- **Source build**: 2.56.5 with Apple Silicon fixes
- **Homebrew**: 2.57.3 (latest with all recent patches)
- **Both fail identically** with same segmentation faults

## Conclusion
Despite extensive debugging across multiple installation methods, build configurations, system modifications, and permission changes, Intel's RealSense SDK demonstrates **fundamental incompatibility** with Apple Silicon Macs running current macOS versions. The consistent failure pattern across different versions and installation approaches indicates this is not a configuration issue but a **systemic compatibility problem** that may be unresolvable without significant changes from either Intel or Apple.

## Recommendations for Other Users
1. **Intel MacBook Pro 2019**: x86 architecture may have better compatibility
2. **Alternative depth cameras**: Azure Kinect, OAK-D cameras with documented macOS support
3. **Virtual machines**: Windows/Linux VMs for RealSense development
4. **Network streaming**: Use working system to capture depth data, stream to Mac for processing

This extensive testing confirms the reports from multiple users experiencing similar issues across GitHub issues #13640, #12890, #11815, and others.

jshaw, 2025-09-19:
Also following up with tests using a intel mbp...

---

Update: Intel MacBook Pro 2019 Testing Results
I tested the same D455C camera on an Intel MacBook Pro 2019 to determine if this is an Apple Silicon-specific issue. Results confirm this affects Intel Macs as well.
Intel Mac Test Results:

Hardware: MacBook Pro 2019 (Intel x86_64)
Installation: Homebrew librealsense 2.57.3
Command: sudo realsense-viewer

Partial Success:
The viewer did start loading (which never happened on M2), showing:

RealSense context initialization succeeded
USB device detection worked
Graphics interface began loading

Same Crash Pattern:
However, it crashed with identical symptoms:
Exception Type: EXC_BAD_ACCESS (SIGSEGV)
Exception Codes: KERN_INVALID_ADDRESS at 0x0000000000000018
Crashed in: IOHIDDeviceOpen() during HID device initialization
Key Findings:

Cross-platform issue: Both Apple Silicon M2 and Intel x86_64 exhibit identical crash patterns
USB vs HID access: Intel Mac can detect camera via USB but crashes when accessing HID interface (motion sensors)
macOS-wide problem: This isn't architecture-specific but affects Intel's SDK across all Mac hardware

Conclusion:
The Intel Mac testing proves this is a fundamental macOS compatibility issue with Intel's RealSense SDK, not an Apple Silicon limitation. Even Intel's historically better-supported x86_64 architecture fails with the same HID access restrictions, confirming this is a systemic problem requiring Intel/Apple-level resolution.

MartyG-RealSense, 2025-09-20:
Hi @jshaw  D455C is not an official RealSense camera model name.  It is likely that it is a version of the D455 sold by the company **DotProduct** that has had additional calibration applied to it.

https://www.point3d.com/product/intel-realsense-d455-kit/

As the camera is detected as a D455 though, it is likely that the only difference from a standard D455 is the custom calibration applied by DotProduct and that the issue is primarily related to Mac Apple Silicon M and MacOS.

RealSense cameras have been able to be used successfully with Apple Silicon M Macs, though often the installation process has issues that require discussion with other RealSense Mac users to resolve.  RealSense does not officially support the Mac platform, though does introduce fixes into the RealSense SDK when possible in order to improve Mac compatibility.

The most stable M-compatible MacOS version was 12 (Monterey), and the company **Lightbuzz** released a RealSense installation guide for Monterey and Apple Silicon Macs.

https://lightbuzz.com/realsense-macos/

In subsequent MacOS versions after Monterey though (Ventura and Sonoma), compatibility began decreasing.  It is currently thought that it is related to changes that Apple made to DriverKit, as described at https://github.com/IntelRealSense/librealsense/issues/9916#issuecomment-2988188804

The sharing of your extensive test results is very appreciated and I will share it with my RealSense colleagues.

MartyG-RealSense, 2025-09-28:
Hi again @jshaw  I wanted to update you that after discussion of your Mac report with my RealSense colleagues, it is still believed that the main core of the problems is with Apple's DriverKit.