realsenseai/librealsense #14306: D435i cannot stream imu, even without rgb and depth

Go2SchooI, 2025-09-22:
I am able to get rgb and depth stream. But when I tried get imu stream, it says "RuntimeError: Frame didn't arrive within 5000"
I think this maybe because I tried get all the data from one stream so I tried code from this https://github.com/IntelRealSense/librealsense/issues/5628, but it is still "RuntimeError: Frame didn't arrive within 5000":

"""
Author: Ed Murphy
File: realsense435i.py
Date: April 14 2019
Notes: Donkeycar part for the Intel Realsense depth cameras D435 and D435i.
"""
import argparse
import time
import logging
import sys

import numpy as np
import pyrealsense2 as rs

#
# NOTE: Jetson Nano users should clone the Jetson Hacks project
#       https://github.com/JetsonHacksNano/installLibrealsense
#       and _build librealsense from source_ in order to get the python bindings.
#       ./buildLibrealsense.sh
#

#
# The Realsense D435 will not output images with arbitrarily chosen dimensions.
# The dimensions must be from the allowed list.  I've chosen a reasonable
# resolution that I know is supported by the camera.
# If the part is initialized with different sizes, then opencv
# will be used to resize the image before it is returned.
#
WIDTH = 424
HEIGHT = 240
CHANNELS = 3

class RealSense435i(object):
    """
    Donkeycar part for the Intel Realsense depth cameras D435 and D435i.
    The Intel Realsense D435i camera is a device which uses an imu, twin fisheye cameras,
    and an Movidius chip to stream a depth map along with an rgb image and optionally,
    accelerometer and gyro data (the 'i' variant has an IMU, the non-i variant does not)
    NOTE: this ALWAYS captures 424 pixels wide x 240 pixels high x RGB at 60fps.
          If an image width, height or depth are passed with different values,
          the the outputs will be scaled to the requested size on the way out.
    """

    def __init__(self, width = WIDTH, height = HEIGHT, channels = CHANNELS, enable_rgb=True, enable_depth=True, enable_imu=False, device_id = None):
        self.device_id = device_id  # "923322071108" # serial number of device to use or None to use default
        self.enable_imu = enable_imu
        self.enable_rgb = enable_rgb
        self.enable_depth = enable_depth

        self.width = width
        self.height = height
        self.channels = channels
        self.resize = (width != WIDTH) or (height != height) or (channels != CHANNELS)
        if self.resize:
            print("The output images will be resized from {} to {}.  This requires opencv.".format((WIDTH, HEIGHT, CHANNELS), (self.width, self.height, self.channels)))

        # Configure streams
        self.imu_pipeline = None
        if self.enable_imu:
            self.imu_pipeline = rs.pipeline()
            imu_config = rs.config()
            if None != self.device_id:
                imu_config.enable_device(self.device_id)
            imu_config.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 100)  # acceleration
            imu_config.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 200)  # gyroscope
            imu_profile = self.imu_pipeline.start(imu_config)
            # eat some frames to allow imu to settle
[... 222 lines cut ...]
                        depth_colormap = cv2.cvtColor(cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET), cv2.COLOR_RGB2GRAY) if enable_depth else None

                    # Stack both images horizontally
                    images = None
                    if enable_rgb:
                        images = np.hstack((color_image, depth_colormap)) if enable_depth else color_image
                    elif enable_depth:
                        images = depth_colormap

                    if images is not None:
                        cv2.imshow('RealSense', images)

                # Press esc or 'q' to close the image window
                key = cv2.waitKey(1)
                if key & 0xFF == ord('q') or key == 27:
                    cv2.destroyAllWindows()
                    break
            if profile_frames > 0:
                if frame_count == profile_frames:
                    print("Aquired {} frames in {} seconds for {} fps".format(str(frame_count), str(frame_time - start_time), str(frame_count / (frame_time - start_time))))
                    break
            else:
                time.sleep(0.05)
    finally:
        camera.shutdown()

MartyG-RealSense, 2025-09-22:
Hi @Go2SchooI  Does the error still occur if you store the frames in the computer's memory using the **keep()** instruction by defining a **frames** variable for wait_for_frames like in the Python code below?

```
frames = self.pipeline.wait_for_frames()
frames.keep()
```

Go2SchooI, 2025-09-22:
> Hi [@Go2SchooI](https://github.com/Go2SchooI) Does the error still occur if you store the frames in the computer's memory using the **keep()** instruction by defining a **frames** variable for [...]

Hi, thank you very much for the suggestion. My apologies for not posting the full code initially; I realize that may have caused some confusion. Here is the complete code for context, which I copied from https://github.com/IntelRealSense/librealsense/issues/5628:
```

"""
Author: Ed Murphy
File: realsense435i.py
Date: April 14 2019
Notes: Donkeycar part for the Intel Realsense depth cameras D435 and D435i.
"""
import argparse
import time
import logging
import sys

import numpy as np
import pyrealsense2 as rs

#
# NOTE: Jetson Nano users should clone the Jetson Hacks project
#       https://github.com/JetsonHacksNano/installLibrealsense
#       and _build librealsense from source_ in order to get the python bindings.
#       ./buildLibrealsense.sh
#

#
# The Realsense D435 will not output images with arbitrarily chosen dimensions.
# The dimensions must be from the allowed list.  I've chosen a reasonable
# resolution that I know is supported by the camera.
# If the part is initialized with different sizes, then opencv
# will be used to resize the image before it is returned.
#
WIDTH = 424
HEIGHT = 240
CHANNELS = 3

class RealSense435i(object):
    """
    Donkeycar part for the Intel Realsense depth cameras D435 and D435i.
    The Intel Realsense D435i camera is a device which uses an imu, twin fisheye cameras,
    and an Movidius chip to stream a depth map along with an rgb image and optionally,
    accelerometer and gyro data (the 'i' variant has an IMU, the non-i variant does not)
    NOTE: this ALWAYS captures 424 pixels wide x 240 pixels high x RGB at 60fps.
          If an image width, height or depth are passed with different values,
          the the outputs will be scaled to the requested size on the way out.
    """

    def __init__(self, width = WIDTH, height = HEIGHT, channels = CHANNELS, enable_rgb=True, enable_depth=True, enable_imu=False, device_id = None):
        self.device_id = device_id  # "923322071108" # serial number of device to use or None to use default
        self.enable_imu = enable_imu
        self.enable_rgb = enable_rgb
        self.enable_depth = enable_depth

        self.width = width
        self.height = height
        self.channels = channels
        self.resize = (width != WIDTH) or (height != height) or (channels != CHANNELS)
        if self.resize:
            print("The output images will be resized from {} to {}.  This requires opencv.".format((WIDTH, HEIGHT, CHANNELS), (self.width, self.height, self.channels)))

        # Configure streams
        self.imu_pipeline = None
        if self.enable_imu:
            self.imu_pipeline = rs.pipeline()
            imu_config = rs.config()
            if None != self.device_id:
                imu_config.enable_device(self.device_id)
            imu_config.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 100)  # acceleration
            imu_config.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 200)  # gyroscope
[... 229 lines cut ...]
                        images = np.hstack((color_image, depth_colormap)) if enable_depth else color_image
                    elif enable_depth:
                        images = depth_colormap

                    if images is not None:
                        cv2.imshow('RealSense', images)

                # Press esc or 'q' to close the image window
                key = cv2.waitKey(1)
                if key & 0xFF == ord('q') or key == 27:
                    cv2.destroyAllWindows()
                    break
            if profile_frames > 0:
                if frame_count == profile_frames:
                    print("Aquired {} frames in {} seconds for {} fps".format(str(frame_count), str(frame_time - start_time), str(frame_count / (frame_time - start_time))))
                    break
            else:
                time.sleep(0.05)
    finally:
        camera.shutdown()

```

Following up on your suggestion, I tried adding frames.keep(), but unfortunately, the result is the same. The RuntimeError: Frame didn't arrive within 5000 still occurs. In fact, the error happens even when I only enable the IMU stream (with RGB and depth disabled). 
As you can see from the full code, it seems to be failing during the object's initialization in the __init__ method, specifically at the self.imu_pipeline.wait_for_frames() line.

MartyG-RealSense, 2025-09-22:
You could next test some of the IMU-only Python scripts at https://github.com/IntelRealSense/librealsense/issues/8492 to test whether the issue is with your script specifically or any Python script that tries to access the IMU.

You could also try running your original test script in **sudo** admin permissions mode to see whether it makes a difference.

`sudo python3 realsense435i.py`

If realsense-viewer continues to work whilst Python scripts still produce the `Frame didn't arrive within 5000` error - which has happened for some RealSense users - then some users have found that a complete fresh reinstallation of librealsense and the Python wrapper resolves the issue.

Go2SchooI, 2025-09-22:
> You could next test some of the IMU-only Python scripts at [#8492](https://github.com/IntelRealSense/librealsense/issues/8492) to test whether the issue is with your script specifically or any [...]

Hi, thanks for the suggestions.

Following your advice, I created a fresh environment. In this environment, I can now successfully stream IMU data using the test scripts in both https://github.com/IntelRealSense/librealsense/issues/8492 and https://github.com/IntelRealSense/librealsense/issues/5628. The original issue was very likely caused by a package conflict in my old environment.

However, a new problem has emerged. While I can stream imu data or rgb and depth data individually, I cannot stream them simultaneously using the script from https://github.com/IntelRealSense/librealsense/issues/5628.

When I try to initialize both pipeline, the program fails. The traceback shows that the first pipeline (for IMU) seems to start successfully, but the second pipeline (for rgb and depth) fails with a RuntimeError when it tries to get frames. Here is the exact terminal output from the new environment:
```
The output images will be resized from (424, 240, 3) to (212, 120, 3).  This requires opencv.
Depth Scale is:  0.0010000000474974513
Traceback (most recent call last):
  File "realsense435i.py", line 259, in <module>
    camera = RealSense435i(
  File "realsense435i.py", line 103, in __init__
    self.pipeline.wait_for_frames()
RuntimeError: Frame didn't arrive within 5000

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "realsense435i.py", line 317, in <module>
    camera.shutdown()
NameError: name 'camera' is not defined

```

Go2SchooI, 2025-09-25:
> If your camera was manufactured before mid 2022 then Accel should be 63. If the camera is newer than mid 2022 then Accel should be 100.
> 
> This is because the IMU component used in RealSense cameras was changed after mid 2022 and 63 for Accel was no longer supported. Requesting 63 for Accel in newer cameras will result in **RuntimeError: Couldn't resolve requests**

My apologies for the response late as we had a typhoon here, which prevented me from doing any further testing until now. Thank you for the information regarding the accel frequency. I'm not certain when my d435i was made, but using 100 Hz works correctly in both realsense-viewer and python scripts(single pipeline), so I believe that **configuration is correct**.

Therefore, this doesn't seem to be the reason for the issue, which is that I'm unable to read data from two pipelines simultaneously. What I find most confusing is that I can use realsense-viewer to stream data from imu, rgb, and depth all at the same time without any problems. This should prove that the **hardware(including the d435i camera, the USB port, and the cable) is functioning correctly**.

However, I still see no improvement when **using the most stable script** you provided; it still results in timeout errors. I'm quite confused about what the cause could be at this point, since the hardware seems fine. Any further ideas would be greatly appreciated.

Go2SchooI, 2025-09-26:
> In [#5628](https://github.com/IntelRealSense/librealsense/issues/5628) there were several versions of Ezward's script as they made improvements to it and the final, most stable version is below. Is that the code your script is based on, please?
> 
> ```
> ## License: Apache 2.0. See LICENSE file in root directory.
> ## Parts of this code are
> ## Copyright(c) 2015-2017 Intel Corporation. All Rights Reserved.
> 
> ##################################################
> ##      configurable realsense viewer           ##
> ##################################################
> 
> import pyrealsense2 as rs
> import numpy as np
> import cv2
> import time
> 
> #
> # NOTE: it appears that imu, rgb and depth cannot all be running simultaneously.
> #       Any two of those 3 are fine, but not all three: causes timeout on wait_for_frames()
> #
> device_id = None  # "923322071108" # serial number of device to use or None to use default
> enable_imu = True
> enable_rgb = True
> enable_depth = True
> # TODO: enable_pose
> # TODO: enable_ir_stereo
> 
> 
> # Configure streams
> if enable_imu:
>     imu_pipeline = rs.pipeline()
>     imu_config = rs.config()
>     if None != device_id:
>         imu_config.enable_device(device_id)
>     imu_config.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 63) # acceleration
>     imu_config.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 200)  # gyroscope
>     imu_profile = imu_pipeline.start(imu_config)
> 
> 
> if enable_depth or enable_rgb:
>     pipeline = rs.pipeline()
>     config = rs.config()
> 
>     # if we are provided with a specific device, then enable it
>     if None != device_id:
>         config.enable_device(device_id)
> 
>     if enable_depth:
>         config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 60)  # depth
> 
>     if enable_rgb:
>         config.enable_stream(rs.stream.color, 424, 240, rs.format.bgr8, 60)  # rgb
> 
>     # Start streaming
>     profile = pipeline.start(config)
> 
>     # Getting the depth sensor's depth scale (see rs-align example for explanation)
>     if enable_depth:
>         depth_sensor = profile.get_device().first_depth_sensor()
>         depth_scale = depth_sensor.get_depth_scale()
>         print("Depth Scale is: ", depth_scale)
>         if enable_depth:
>             # Create an align object
>             # rs.align allows us to perform alignment of depth frames to others frames
>             # The "align_to" is the stream type to which we plan to align depth frames.
>             align_to = rs.stream.color
>             align = rs.align(align_to)
> 
> try:
>     frame_count = 0
[... 110 lines cut ...]
libstdcxx-ng               11.2.0           h1234567_1
libxcb                     1.17.0           h9b100fa_0
libzlib                    1.3.1            hb25bd0a_0
ncurses                    6.5              h7934f7d_0
numpy                      1.24.4           pypi_0           pypi
opencv-python              4.12.0.88        pypi_0           pypi
openssl                    3.0.17           h5eee18b_0
pip                        24.2             py38h06a4308_0
pthread-stubs              0.3              h0ce48e5_1
pyrealsense2               2.55.1.6486      pypi_0           pypi
python                     3.8.20           he870216_0
readline                   8.3              hc2a1206_0
setuptools                 75.1.0           py38h06a4308_0
sqlite                     3.50.2           hb25bd0a_1
tk                         8.6.15           h54e0aa7_0
wheel                      0.44.0           py38h06a4308_0
xorg-libx11                1.8.12           h9b100fa_1
xorg-libxau                1.0.12           h9b100fa_0
xorg-libxdmcp              1.1.5            h9b100fa_0
xorg-xorgproto             2024.1           h5eee18b_1
xz                         5.6.4            h5eee18b_1
zlib                       1.3.1            hb25bd0a_0
```

Since the Python code and the conda package versions are virtually identical, could this issue be related to linux-specific factors like kernel patches, USB driver permissions, or the way `librealsense` was installed and interacts with the ubuntu kernel? Any insights you have on this would be extremely helpful.

MartyG-RealSense, 2025-09-26:
You are very welcome.  I'm pleased to hear that you were successful.  Thanks very much for the update!

closed 2025-10-06 by MartyG-RealSense