realsenseai/librealsense #14325: D435 Realsense GUI Not working in macbook m4 sequoia version 15.6

sidb-nyneos, 2025-09-30:
Hi all,

I am trying to run Intel RealSense D435 on a MacBook M4 with macOS Sequoia 15.6 (Apple Silicon). While I can build librealsense and use Python scripts, the RealSense Viewer GUI does not work.

⚙️ Environment

Device: Intel RealSense D435

macOS: Sequoia 15.6 (Apple Silicon, M4 chip)

Python: 3.13.7 (Homebrew), 3.9.6 (system)

librealsense: built from master

Tools installed: CMake, libusb, pkg-config

✅ What Works

rs-enumerate-devices detects the camera (after fixing USB permissions).

Python bindings (pyrealsense2) work inside a virtualenv after manual symlink and DYLD_LIBRARY_PATH fix.

I can capture frames (color + depth) using Python.

❌ What’s Broken

RealSense Viewer (C++ GUI)

Fails with OpenGL/ImGui shader errors:

ERROR: ImGui_ImplOpenGL3_CreateDeviceObjects: failed to compile shader!
With GLSL: #version 150 / #version 330 / #version 410 core
version not supported on macOS

macOS only supports OpenGL 4.1 Core, so shaders beyond/below that fail.

USB Access Issues

Occasionally:

failed to claim usb interface: RS2_USB_STATUS_ACCESS
failed to set power state

Works when running as root, but unstable.

Python Binding Path Issues

pyrealsense2 installs into /opt/homebrew/lib/python3.13/site-packages/ but looks for librealsense2.2.56.dylib in the wrong place.

Fixed by exporting DYLD_LIBRARY_PATH=/usr/local/lib.

No GUI Alternative

Since the official viewer crashes, I attempted a PyQt-based custom GUI, but hit compatibility issues with PyQt5/PyQt6.

📌 Questions for the Community

Has anyone successfully run RealSense Viewer GUI on Apple Silicon macOS (Sequoia or Sonoma)?

What’s the correct way to patch ImGui/OpenGL GLSL for macOS 4.1 Core support?

Is there an official or community workaround for the USB/libusb power state issue on macOS?

Any help, patches, or workarounds from the community would be greatly appreciated 🙏

Thanks,
Sid

MartyG-RealSense, 2025-10-01:
Hi @sidb-nyneos  The RealSense team hopes to improve MacOS compatibility over time for Apple Silicon 'M' processors and recent MacOS versions (MacOS 12 Monterey has been the most stable of the M-supporting MacOS's, with stability declining in subsequent MacOS releases).  It is thought by the RealSense developer team that the issues are related to changes that Apple made to **DriverKit**.

Detailed feedback like yours from the RealSense MacOS user community is very important to RealSense as it helps to guide the development team towards possible future compatibility improvements.  The community is also the best source of advice for resolving Mac Apple Silicon M issues.

Have you tried launching the RealSense Viewer in sudo admin permissions mode with `sudo realsense-viewer`

If that does not make a difference, when you launch realsense-viewer are you able to access any of the options despite the error or does the program just crash and exit completely?

If you believe that your graphics issue in realsense-viewer is related to GLSL, version 2.57.2 beta and newer disables GLSL by default.  Previously GLSL was enabled by default, meaning that if there was a GLSL conflict that crashed and exited the Viewer then it was not possible to fix it by entering the Viewer settings to turn GLSL off.

The betas do not have PyPi pyrealsense2 wheel packages though, so it is only possible to build pyrealsense2 from source code in these beta versions as `pip install pyrealsense2` will not work.

matbeedotcom, 2025-10-08:
Basically, realsense cameras will never work on mac. There's a core issue with what looks like libusb

MartyG-RealSense, 2025-10-26:
Hi @sidb-nyneos  Bearing in mind the advice provided in the comments above, do you require further assistance with this case please?  Thanks!

closed 2025-10-27 by MartyG-RealSense